GeminiDRSoftware/DRAGONS #41: cythonize on gempy/library produces invalid output with Python 3.5.6

marc-white, 2019-10-24:
I've started checking our GHOSTDR development (https://github.com/ANU-RSAA/GHOSTDR) against Python 3.5, given there are Python 3+ syntax and dependencies now within DRAGONS.

I've been doing this by running our test suite in an environment similar to the original Python 2.7 `geminidev` environment that our install instructions required, by updated to Python 3.5.6 (and with relevant other package updates). I'm currently getting the following error, having run the `cythonize` command within `DRAGONS/gempy/library` as instructed:

```
ImportError while importing test module '/Users/marc/Documents/ghost/git/GHOSTDR/ghostdr/ghost/test/test_ghost_spect.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
ghostdr/ghost/test/test_ghost_spect.py:32: in <module>
    from ghostdr.ghost.primitives_ghost_spect import GHOSTSpect
ghostdr/ghost/primitives_ghost_spect.py:31: in <module>
    from gempy.mosaic.mosaicAD import MosaicAD
E   ImportError: bad magic number in 'gempy.mosaic': b'\x03\xf3\r\n'
```

Given mosiacAD doesn't appear to be in the repository, it appears to be created by the `cythonize` command instead.

chris-simpson, 2019-10-24:
"Bad magic number" can be raised when importing a compiled `.pyc` file that's incompatible with the current environment. This will presumably happen here because MosaicAD has been gone for a while so there's no uncompiled version to replace it with. Go to the top level of your package and do

rm -rf `find . -name "*.pyc"

I imagine you'll then get an ImportError as it fails to find MosaicAD, which I don't think you need any more, as per my email of October 14. If the top-level tileArrays() primitive doesn't work for you, let me know and I can work with you to resolve any issues, but it should be fine if the individual extensions return the correct values for the various "section" descriptors.

rcardenes, 2019-10-24:
I agree with @chris-simpson's assessment. Python will ignore the incompatible `.pyc` (and updated it or create a `__pycache__`) as long as the matching `.py` is available.

marc-white, 2019-10-25:
Thanks @rcardenes and @chris-simpson, that seems to have fixed the issue.

Is there a way that process can be automated within DRAGONS? E.g. adding a command to purge all .pyc files as part of the cythonize command?

chris-simpson, 2019-10-25:
I don't think it's connected to cythonize. I had a spate of Bad Magic Number errors when I was having to switch between the DRAGONS spectroscopy branch (py3 only) and bug fixes in master (which needed to be py2/3 compatible). So there would be orphaned py3 .pyc files that would try to get imported. I presume you were running cythonize because of a python version change.

https://stackoverflow.com/questions/514371/whats-the-bad-magic-number-error

closed 2020-11-26 by chris-simpson