GeminiDRSoftware/DRAGONS #112: Tox environment with Python 3.6 does not run

b1quint, 2021-01-16:
I had to run some tests locally to verify some regression tests. I didn't remember that Py37 is now a requirement for DRAGONS so I ran the tests using one of the py36 environments. I got the following error: 

[traceback.txt](https://github.com/GeminiDRSoftware/DRAGONS/files/5824646/traceback.txt)

My first question is if DRAGONS should run with Py36 or not. I remember some discussions but I don't remember the outcome. If yes, we have to find a solution to the error above. If not, we have to remove py36 from tox.

saimn, 2021-01-18:
On master we require Astropy 4.2 which itself requires Python 3.7

saimn, 2021-01-18:
(Python 3.6 only gets security fixes now, and its support ends at the end of the 2021).

b1quint, 2021-01-19:
Ok, so this is not a but. I will remove the label. 

I kind of recall some discussions related to that but I could not find them on Slack. Thank you for the clarification.

closed 2021-01-19 by b1quint