GeminiDRSoftware/DRAGONS #243: Unable to install following instructions in README

DBerke, 2021-04-28:
I attempted to install DRAGONS today using the command given in the README file, `$ conda create -n geminiconda python=3.6 gemini stsci`, which failed with the following error:

```
$ conda create -n geminiconda python=3.6 gemini stsci
Collecting package metadata (current_repodata.json): done
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - stsci
  - gemini

Current channels:

  - https://repo.anaconda.com/pkgs/main/linux-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/linux-64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```
Following the instructions in the error message I went to https://anaconda.org and searched for the missing packages, but couldn't find either of them unambiguously: I found packages named `stsci_tools` and `stsci_distutils` but none of the (few) hits for `gemini` seemed to be astronomy related. I'm using a fresh Conda install, version 4.10.0 if that's relevant.

olyoberdorf, 2021-04-28:
Hi Dr. Berke,

The README may need to be updated, but in the meantime does adding these repos solve your issue?

conda config --add channels http://ssb.stsci.edu/astroconda
conda config --add channels http://astroconda.gemini.edu/public

DBerke, 2021-04-28:
Yes, adding those repos solved it and I was able to proceed without further problems. Thanks. That may be all that's necessary to add to the README. I'll go ahead and close this issue.

(And honesty compels me to mention that I still have a few months left until I [hopefully!] earn the title of Dr., but thank you.)

olyoberdorf, 2021-04-28:
Congrats in advance then :)

reopening and assigning to myself as a reminder to do the update

KathleenLabrie, 2021-05-06:
That was my mistake.   I fixed it and added the info about the two channels.  Thanks for reaching out and reporting the issue.

closed 2021-05-06 by KathleenLabrie