GeminiDRSoftware/DRAGONS #246: trace regions keyboard shortcuts do not work.

olyoberdorf, 2021-05-06:
per KL:

Another issue, the regions keyboard shortcuts do not work.

KathleenLabrie, 2021-05-06:
Actually, they do work.  With an insane delay.   Which reinforce my impression that `traceApertures` has got much slower in a recent update.  Anyway, that will be a different ticket probably.  Back to this one.

I'm launching in interactive mode.  There are 10 apertures (might be relevant).  First it takes forever to pop up, it is doing something first.   On the plot, I type "r", then I wait, and wait, several seconds.  The instruction menu updates (the "E" option disappear, that's my queue.)  Now I can define a region with the cursor.   I press "r" again, and wait, wait, several seconds, and the regions is finally defined, the "E" in the instructions reappears and the plot gets updated with the new fit.

KathleenLabrie, 2021-05-06:
When there's only one aperture, there is no lag.   Also the stuff it does before the pop up is super quick, <2 sec, just manual timer, too fast for me.

My gut feeling is that the code does something silly when there are multiple apertures.

KathleenLabrie, 2021-05-06:
I'm convinced that this is related to #253

olyoberdorf, 2021-05-06:
Yeah these are "working" but I see the slowness with 11 apertures in Chrome.  I'll work on 253 first as it will likely fix this as well.

olyoberdorf, 2021-05-21:
fixed the aperture slowness issue, closing this as related

closed 2021-05-21 by olyoberdorf