GeminiDRSoftware/DRAGONS #252: Model's domain inconsistent when running trace interactive on certain data

olyoberdorf, 2021-05-06:
from KL:

traceAperture interactive is causing trouble.   When I run it in non-interactive move, extract is clean.  When I run it in interactive mode and just click Accept, no changes, extract1DSpectra has those warnings.  I do not know what those warnings mean:
   PRIMITIVE: extract1DSpectra
   ---------------------------
   Extracting 10 spectra from S20190206S0108_2D.fits extension 1
       Extracting spectrum from aperture 1
   WARNING - Model's domain is inconsistent with image size. Results may be incorrect.
       Extracting spectrum from aperture 2
   WARNING - Model's domain is inconsistent with image size. Results may be incorrect.
       Extracting spectrum from aperture 3
   WARNING - Model's domain is inconsistent with image size. Results may be incorrect.
       Extracting spectrum from aperture 4
   WARNING - Model's domain is inconsistent with image size. Results may be incorrect.
       Extracting spectrum from aperture 5
   WARNING - Model's domain is inconsistent with image size. Results may be incorrect.
       Extracting spectrum from aperture 6
   WARNING - Model's domain is inconsistent with image size. Results may be incorrect.
       Extracting spectrum from aperture 7
   WARNING - Model's domain is inconsistent with image size. Results may be incorrect.
       Extracting spectrum from aperture 8
   WARNING - Model's domain is inconsistent with image size. Results may be incorrect.
   WARNING - Aperture lower limit is greater than zero.
       Extracting spectrum from aperture 9
   WARNING - Model's domain is inconsistent with image size. Results may be incorrect.
       Extracting spectrum from aperture 10
   WARNING - Model's domain is inconsistent with image size. Results may be incorrect.
Note that I did not see those warning when I ran trace interactively on the standard for which there was only one file and one aperture.

olyoberdorf, 2021-05-06:
may be related to #247

KathleenLabrie, 2021-05-06:
It's the same.  Maybe close the other one as this one has a bit more info.   

I'll set up a Gdrive with the data.  Whoever picks this one up, get in touch with me and I'll give you the link to the data.

olyoberdorf, 2021-05-10:
domain was not being passed down to the fit and it was inferring min:max off x