GeminiDRSoftware/DRAGONS #253: traceApertures gets very slow when there are several apertures

KathleenLabrie, 2021-05-06:
<master 8533796e8b3>

traceApertures (regardless of interactivity) gets very slow in a non-linear way when there are multiple apertures.

Running in non-interactive mode, I manually timed the following:
* 1 aperture:  < 2 secs  => < 2 sec / aperture
* 10 apertures:  80 secs  =>  8 sec / aperture

Whatever causes this affects the interactive tool's responsiveness too.  See issue #246.

(On my todo, is to time it using the release/3.0.x branch.  I'll post the results later.)

olyoberdorf, 2021-05-06:
I'm going to try clearing the data in the figures when they are not the active tab.

KathleenLabrie, 2021-05-06:
Using the release/3.0.x branch shows a similar behavior.

* 6 apertures:  40 seconds =>  6.5 sec / aperture

KathleenLabrie, 2021-05-06:
This behavior probably has absolutely nothing to do with the interactive stuff.  The code in `release/3.0.x` does not have any interactive code.   It looks like a recalculation loop is going bad causing an exponential rise in the time needed.

chris-simpson, 2021-05-13:
This is not related to the number of apertures, but the size of the image. The examples here, the 1-aperture image is a Central Spectrum, and the 10-aperture one a Full Frame, which is 4 times longer in the slit direction, and therefore that's the cause of the slower performance.

The performance of the code can probably be improved but this isn't a bug.

olyoberdorf, 2021-05-13:
In that case, I'll see if I can do something in the interactive code for the related issue (that the UI is quite slow in this example)

olyoberdorf, 2021-05-21:
This has been fixes as well as possible by dynamically removing and adding elements to the DOM via bokeh in repsonse to tab selections

closed 2021-05-21 by olyoberdorf