GeminiDRSoftware/DRAGONS #374: reduce should check that darks used for processing flats have the same exposure time

DBerke, 2022-07-27:
Currently (2022-07-26), the `reduce` command, when used on flats which require darks (such as F2), allows you to pass a list of darks without checking that the exposure times of both groups of frames match. Somewhere along the way it should probably check that they do.

olyoberdorf, 2022-07-27:
This is not as straightforward an issue as one would think, so I'll toss some thoughts here.

The vast majority of the time, users would just add their cals to caldb and they would match.  To me, the primary use of the command line specified cals is when that mechanism doesn't work for some reason.  Perhaps the user disagrees with one of the matching requirements or knows a keyword is wrong and doesn't want to fuss with modifying it.

If we start enforcing cal matching rules on the user specified cals, that override scenario doesn't work without maybe an additional force parameter or such.

On the other hand, if the idea is to warn the user, that could be done but there are caveats.  It would have to integrate something like the mechanism I used in the calcheck script.  So it would be building an in-memory caldb and using that to check that they match.  If they don't, it could log a warning and that may be useful.  It could also print a list of the matching terms it is checking but that is done as a best effort and won't work for complex queries.  I'd be happy to do this, but @chris-simpson would need to weigh in on the merit of the extra checking vs the complexity this introduces to get it.

chris-simpson, 2022-07-27:
I think you misunderstand. This refers specifically to the recipes that make flatfields and can take a list of raw lamp frames and a list of raw darks to subtract. Each set is stacked and then the darks are subtraction from the lamps. Just like when we stack darks we check that they all have the same exposure time (I think we do anyway), we should be checking that somebody isn't accidentally sending a list of long-exposure darks (which would be appropriate for their science) instead of the short-exposure darks (for the flats).

We could do this via a warning but an error is fine because the user can create a processed_dark and then subtract that from the flats by passing it as a user_cal.

olyoberdorf, 2022-07-27:
ooh duh, yes completely jumped the shark.