GeminiDRSoftware/DRAGONS #379: Add +/- buttons option for textsliders in interactive builder

olyoberdorf, 2022-08-11:
Daniel suggested we have +/- buttons as an option when building the text sliders.

DBerke, 2022-08-11:
It'd be nice as a way to step through things; if you've got a thousand rows of data, the slider often has a minimum step size > 1, and it'd be easier than having to type in a number in the text box at each step. They might not be needed everywhere, but they shouldn't take up too much space and are a pretty common element, so shouldn't be too difficult to implement I'd imagine.

olyoberdorf, 2022-08-11:
Oh, I used step size for the branch.  Maybe for integer base sliders step size should always be 1.

olyoberdorf, 2022-08-11:
Actually, do you have an example of this?  The code seems to already use stepsize of 1 for integer values.  i.e. what primitive/field?

DBerke, 2022-08-18:
Sorry, that wasn't really clear. It's not so much that the slider actually has a minimum step size > 1, as that if you've got N rows of data and a slider that's < N pixels wide, you can't really move it in increments of ±1 (so I don't really have an example). Though even if you could, it's still fiddly to grab the little slider and move it by exactly one step each time you want to move it. Basically, in my view:

- sliders are great for rapidly jumping around where you don't care what position you land in. (You're initially exploring the data.)
- number entry boxes are great for rapidly getting to an exact position you do care about. (You want to check something in a specific part of the data.)
- spinner boxes are great for stepping through data easily. (You're more methodically exploring or investigating [part of] the data.)

All three have different strengths and weaknesses, but complement each other nicely, so I think having all three would provide the greatest range of use for users.