GeminiDRSoftware/DRAGONS #385: measureIQ was broken during last refactor - hard crash when no sources found

KathleenLabrie, 2022-09-16:
Correct behavior is to recognize that there were no good sources found and just move on.  We can't have the QAP hard crash like that if an image has no visible sources.   This problem is found in `master` and `release/3.1.x`.   `release/3.0.x` behaves the correct way and because it is an AO observation uses info from the header I believe.   

**Master and release/3.1.x**

`reduce -r measureIQ N20131215S0156.fits`

```   
PRIMITIVE: measureIQ
   --------------------
   WARNING - This is an AO observation, the AO-estimated seeing will be used for the IQ band calculation
   WARNING - No OBJCATs found on input. Has detectSources() been run?
ERROR - Traceback (most recent call last):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/reduction/coreReduce.py", line 238, in runr
ERROR -     primitive_as_recipe()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 364, in gn
ERROR -     ret_value = fn(pobj, adinputs=adinputs, **fnargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 293, in gn
ERROR -     ret_value = fn(pobj, **kwargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 373, in measureIQ
ERROR -     report.report(results)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 1193, in report
ERROR -     if results["strehl"]:
ERROR - KeyError: 'strehl'
ERROR - 'strehl'
ERROR - reduce caught an unhandled exception.
ERROR - 
ERROR - 'strehl'
ERROR - Traceback (most recent call last):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/scripts/reduce", line 87, in main
ERROR -     r_reduce.runr()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/reduction/coreReduce.py", line 238, in runr
ERROR -     primitive_as_recipe()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 364, in gn
ERROR -     ret_value = fn(pobj, adinputs=adinputs, **fnargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 293, in gn
ERROR -     ret_value = fn(pobj, **kwargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 373, in measureIQ
ERROR -     report.report(results)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 1193, in report
ERROR -     if results["strehl"]:
ERROR - KeyError: 'strehl'
ERROR - 
ERROR - Reduce instance aborted.```

**release/3.0.x**

   PRIMITIVE: measureIQ
   --------------------
   WARNING - No OBJCATs found on input. Has detectSources() been run?
   WARNING - This is an AO observation, the AO-estimated seeing will be used for the IQ band calculation
   WARNING - No good sources found in N20131215S0156.fits:1
   
      Filename: N20131215S0156.fits
      --------------------------------------------------------
      (Strehl could not be determined)                        
      Zenith-corrected FWHM (AM 1.07):       (AO) 0.796 arcsec
      IQ range for AO-band:             IQ70 (0.45-0.8 arcsec)
      Requested IQ:                                       IQ70
      --------------------------------------------------------
   
   .
	Wrote N20131215S0156_iqMeasured.fits in output directory

reduce completed successfully.

KathleenLabrie, 2022-09-16:
Another file that crashes in a different way after some processing.   Required flat at https://drive.google.com/drive/folders/1rS4OR7ygKdxdMlqMeWduO5Xo1Ux-_I16?usp=sharing

Custom recipe:
```
def reduce(p):
    p.prepare()
    p.addDQ()
    p.ADUToElectrons()
    p.addVAR(read_noise=True, poisson_noise=True)
    p.flatCorrect(flat='N20131215S0408_flat.fits')
    p.detectSources()
    p.measureIQ()
```

**`master` and `release/3.1.x`**

`reduce -r debug.reduce N20131215S0146.fits`

```
   PRIMITIVE: measureIQ
   --------------------
   WARNING - This is an AO observation, the AO-estimated seeing will be used for the IQ band calculation
ERROR - Reduce received an unhandled exception. Aborting ...
ERROR - Traceback (most recent call last):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/reduction/coreReduce.py", line 246, in runr
ERROR -     recipe(p)
ERROR -   File "/Users/klabrie/data/workspace/dragons/niri/debug.py", line 8, in reduce
ERROR -     p.measureIQ()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 364, in gn
ERROR -     ret_value = fn(pobj, adinputs=adinputs, **fnargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 293, in gn
ERROR -     ret_value = fn(pobj, **kwargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 349, in measureIQ
ERROR -     results = report.calculate_metric(ao_seeing_fn=ao_seeing_fn)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 1147, in calculate_metric
ERROR -     self.calculate_qa_band(zfwhm, zfwhm_std)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 604, in calculate_qa_band
ERROR -     norm_limits = [(l - float(value))/unc for l in limits]
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 604, in <listcomp>
ERROR -     norm_limits = [(l - float(value))/unc for l in limits]
ERROR - TypeError: unsupported operand type(s) for /: 'float' and 'NoneType'
Writing final outputs ...
	Wrote N20131215S0146_sourcesDetected.fits in output directory
ERROR - reduce caught an unhandled exception.
ERROR - 
ERROR - unsupported operand type(s) for /: 'float' and 'NoneType'
ERROR - Traceback (most recent call last):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/scripts/reduce", line 87, in main
ERROR -     r_reduce.runr()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/reduction/coreReduce.py", line 246, in runr
ERROR -     recipe(p)
ERROR -   File "/Users/klabrie/data/workspace/dragons/niri/debug.py", line 8, in reduce
ERROR -     p.measureIQ()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 364, in gn
ERROR -     ret_value = fn(pobj, adinputs=adinputs, **fnargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 293, in gn
ERROR -     ret_value = fn(pobj, **kwargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 349, in measureIQ
ERROR -     results = report.calculate_metric(ao_seeing_fn=ao_seeing_fn)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 1147, in calculate_metric
ERROR -     self.calculate_qa_band(zfwhm, zfwhm_std)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 604, in calculate_qa_band
ERROR -     norm_limits = [(l - float(value))/unc for l in limits]
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 604, in <listcomp>
ERROR -     norm_limits = [(l - float(value))/unc for l in limits]
ERROR - TypeError: unsupported operand type(s) for /: 'float' and 'NoneType'
ERROR - 
ERROR - Reduce instance aborted.
```

**`release/3.0.x`**

```
   PRIMITIVE: measureIQ
   --------------------
   WARNING - This is an AO observation, the AO-estimated seeing will be used for the IQ band calculation
   WARNING - Only one source found. IQ numbers may not be accurate
   
      Filename: N20131215S0146_sourcesDetected.fits
      1 sources used to measure IQ
      --------------------------------------------------------
      FWHM measurement:                 1.138 +/- 0.000 arcsec
      Ellipticity:                             0.104 +/- 0.000
      Strehl ratio:                            0.003 +/- 0.000
      Zenith-corrected FWHM (AM 1.07):       (AO) 0.686 arcsec
      IQ range for AO-band:             IQ70 (0.45-0.8 arcsec)
      Requested IQ:                                       IQ70
                                     WARNING: high ellipticity
      WARNING: single source IQ measurement - no error available
      --------------------------------------------------------
```

KathleenLabrie, 2022-09-16:
All related I guess.  This time measureBG crashes when the nominal zeropoint is missing.  Again, it should just say that there is no nominal zeropoints and continue, not crash.

**`master` and `release/3.1.x`**

`reduce -r measureBG N20131215S0152.fits`

```
   PRIMITIVE: measureBG
   --------------------
   No nominal photometric zeropoint available for N20131215S0152.fits extension 1, filter H
ERROR - Traceback (most recent call last):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/reduction/coreReduce.py", line 238, in runr
ERROR -     primitive_as_recipe()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 364, in gn
ERROR -     ret_value = fn(pobj, adinputs=adinputs, **fnargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 293, in gn
ERROR -     ret_value = fn(pobj, **kwargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 119, in measureBG
ERROR -     report.report(results)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 804, in report
ERROR -     if results['mag'] is not None:
ERROR - KeyError: 'mag'
ERROR - 'mag'
ERROR - reduce caught an unhandled exception.
ERROR - 
ERROR - 'mag'
ERROR - Traceback (most recent call last):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/scripts/reduce", line 87, in main
ERROR -     r_reduce.runr()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/reduction/coreReduce.py", line 238, in runr
ERROR -     primitive_as_recipe()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 364, in gn
ERROR -     ret_value = fn(pobj, adinputs=adinputs, **fnargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 293, in gn
ERROR -     ret_value = fn(pobj, **kwargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 119, in measureBG
ERROR -     report.report(results)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 804, in report
ERROR -     if results['mag'] is not None:
ERROR - KeyError: 'mag'
ERROR - 
ERROR - Reduce instance aborted.

```

**`release/3.0.x`**

```
   PRIMITIVE: measureBG
   --------------------
   No nominal photometric zeropoint available for N20131215S0152.fits:1, filter H
   
      Filename: N20131215S0152.fits
      ----------------------------------------------------------
      Sky level measurement:                      810 +/- 32 ADU
      (BG band could not be determined)                          
      Requested BG:                                        BGAny
      ----------------------------------------------------------
   
   .
	Wrote N20131215S0152_bgMeasured.fits in output directory

```

chris-simpson, 2022-09-23:
These issues have both been fixed in bugfix/qap branch (from master) and tests now exist for these and other issues. The existing measureXX tests had been turned off without my knowledge. I'll cherry-pick the commits into release/3.1.x once Jenkins is happy.

chris-simpson, 2022-09-24:
Now in 3.1.x

KathleenLabrie, 2022-09-28:
Is there something special done for GSAOI? It is failing on S20150528S00112.fits.

strehl, fwhm = results["strehl"], results["fwhm"]
KeyError: 'fwhm'

KathleenLabrie, 2022-09-28:
On the plus side, some other gsaoi crash related to BG measurements have been resolved by this fix.

chris-simpson, 2022-09-28:
Yes, there is something special for GSAOI (remember that document I wrote about estimating the natural seeing from the AO-delivered FWHM). I'll fix and add a test.

chris-simpson, 2022-09-28:
fixed and added test. pushed to `bugfix/qap` and cherry picked into `release/3.1.x`. Jenkins is running.

KathleenLabrie, 2022-09-28:
Thanks.  I think that we're done with this one.  (I ran my test, it runs now.)

chris-simpson, 2022-09-28:
OK, I'm merging `bugfix/qap` into `master` as well then.

closed 2022-09-28 by chris-simpson