GeminiDRSoftware/DRAGONS #386: display in measureIQ does not circle stars if frame not mosaiced

KathleenLabrie, 2022-09-16:
First there was an issue in the refactored `measureIQ` (`master`, `release/3.1.x`) where when an amp was un-illuminated and was not given an OBJCAT, the code would crash on that extension (eg. first and last extension in 12-amp Ham).  That was fixed by checking the content of the report.measurements table, if no columns, just go to next extension.  This is in the loop where the iq star coordinates are compiled.

I mention that just in case the "unrepresented" first and last amp are somehow the source of the problem I describe below.

Problem:  The blue circle identifying the stars selected to measure the seeing do not display unless `mosaicDectector` has been run.  (ie. just 3 extensions, all with OBJCAT)  In `release/3.0.x` the circles are displayed, always, even on raw data with overscan areas still in.

`reduce -r measureIQ -p display=True S20140623S0133_sourcesDetected.fits`  

Pre-processed file : https://drive.google.com/file/d/1OcnXYNwmzGg3uYXRTu623uGNUGa072H6/view?usp=sharing

KathleenLabrie, 2022-09-16:
A clarification.   This seems to happen only for the 12-amp Hamamatsu, so those missing OBJCAT for first and last amp probably have something to do with it.   For an old 3amp EEV, the circles show up on non-mosaiced data.

chris-simpson, 2022-09-24:
Understood and fixed in bugfix/qap and release/3.1.x

KathleenLabrie, 2022-09-28:
Is there something special done for GSAOI?  It is failing on S20150528S00112.fits.

```
strehl, fwhm = results["strehl"], results["fwhm"]
KeyError: 'fwhm'
```

KathleenLabrie, 2022-09-28:
The circles show up now.  One remaining issue related to the logging.  I still get:

`WARNING - No OBJCATs found on input.  Has detectSources() been run?`

This is obviously issued because amp1 and 12 don't have any sources.  Obvious to us, but not obvious at all for users.  There's no information saying that it applies to specific amps.  It reads as if none of the data have OBJCATs.  Is there a way to clarify the message?

chris-simpson, 2022-09-28:
It now says

`WARNING - No OBJCAT(s) found on S20150528S0112.fits EXTVER 1. Has detectSources() been run?`

although it will still say this multiple times if the input has multiple extensions (that's not such a trivial fix).

KathleenLabrie, 2022-09-28:
Thanks.  Better.   One concern.  EXTVER is FITS-centric.  Fine for us since we do only FITS.  However, after all the effort put into trying to avoid too FITSy, I wonder if there would be a better word.  "extension" ?  something else?   I don't know enough about the internals of astrodata to really know if EXTVER is really too FITSy or not.  

Just a thought.  The ticket can be closed, the functionality has been restored.

chris-simpson, 2022-09-28:
Changes made to better reflect how we refer to extensions in the two branches.

closed 2022-09-28 by chris-simpson