GeminiDRSoftware/DRAGONS #387: separateSky fails on GNIRS keyhole image

KathleenLabrie, 2022-09-23:
This happens with `release/3.1.x`.  Might be related to recent changes to how the illumination masks are set now.

```
================================================================================
RECIPE: separateSky
================================================================================
   PRIMITIVE: separateSky
   ----------------------
ERROR - Traceback (most recent call last):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/reduction/coreReduce.py", line 238, in runr
ERROR -     primitive_as_recipe()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 364, in gn
ERROR -     ret_value = fn(pobj, adinputs=adinputs, **fnargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 293, in gn
ERROR -     ret_value = fn(pobj, **kwargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/core/primitives_preprocess.py", line 1288, in separateSky
ERROR -     groups = gt.group_exposures(adinputs, self.inst_lookups, frac_FOV=frac_FOV)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/gempy/gemini/gemini_tools.py", line 2359, in group_exposures
ERROR -     if group.pointing_in_field(ad, fast=True):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/gempy/gemini/gemini_tools.py", line 2248, in pointing_in_field
ERROR -     return self._pointing_in_field(ad, self.group_center,
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gnirs/lookups/FOV.py", line 59, in pointing_in_field
ERROR -     illum = get_illum_mask_filename(ad)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gnirs/lookups/FOV.py", line 106, in get_illum_mask_filename
ERROR -     illum = path.join(maskdb.illumMask_dict[key1, key2])
ERROR - AttributeError: module 'geminidr.gnirs.lookups.maskdb' has no attribute 'illumMask_dict'
ERROR - module 'geminidr.gnirs.lookups.maskdb' has no attribute 'illumMask_dict'
ERROR - reduce caught an unhandled exception.
ERROR - 
ERROR - module 'geminidr.gnirs.lookups.maskdb' has no attribute 'illumMask_dict'
ERROR - Traceback (most recent call last):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/scripts/reduce", line 87, in main
ERROR -     r_reduce.runr()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/reduction/coreReduce.py", line 238, in runr
ERROR -     primitive_as_recipe()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 364, in gn
ERROR -     ret_value = fn(pobj, adinputs=adinputs, **fnargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 293, in gn
ERROR -     ret_value = fn(pobj, **kwargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/core/primitives_preprocess.py", line 1288, in separateSky
ERROR -     groups = gt.group_exposures(adinputs, self.inst_lookups, frac_FOV=frac_FOV)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/gempy/gemini/gemini_tools.py", line 2359, in group_exposures
ERROR -     if group.pointing_in_field(ad, fast=True):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/gempy/gemini/gemini_tools.py", line 2248, in pointing_in_field
ERROR -     return self._pointing_in_field(ad, self.group_center,
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gnirs/lookups/FOV.py", line 59, in pointing_in_field
ERROR -     illum = get_illum_mask_filename(ad)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gnirs/lookups/FOV.py", line 106, in get_illum_mask_filename
ERROR -     illum = path.join(maskdb.illumMask_dict[key1, key2])
ERROR - AttributeError: module 'geminidr.gnirs.lookups.maskdb' has no attribute 'illumMask_dict'
ERROR - 
ERROR - Reduce instance aborted.

reduce exit status: 6
```

**Data available at**:  https://drive.google.com/drive/folders/1LC_aWut174pfPct1G6oOA7K-jDYDyoqv?usp=sharing

To run:
 `reduce -r separateSky *_dqPlaneApplied.fits`

chris-simpson, 2022-09-26:
Oh dear, I've really broken this. There's a fundamental problem (and always has been) with the GNIRS keyhole placement and so I fixed that in master, but within the context of the new way we're handling field overlap calculations, which isn't in 3.1. So can I keep the improved illumination mask placement but make the field overlap calculations work within the old format? This needs some thought.

chris-simpson, 2022-09-27:
I think I have this working now and pushed directly to 3.1.x (no other branches need anything). I'll see about putting together an end-to-end GNIRS imaging reduction test.

KathleenLabrie, 2022-09-28:
Appears to be working fine on all my GNIRS keyhole imaging integration tests.  Thanks.

closed 2022-09-29 by chris-simpson