GeminiDRSoftware/DRAGONS #388: skySubtract gets into an infinite loop

KathleenLabrie, 2022-09-28:
This is a new behavior.  The same call on the same data has been working fine for years, (DRAGONS v2 and DRAGONS v3.0.x)

Data here:  https://drive.google.com/drive/folders/15YgYnOAdvuM_XWwaZWoLZHmuT4hTRkbh?usp=sharing

Call:
`reduce -r skyCorrect @debug.lis -p mask_objects=False reject_method=varclip`

Observed to fail on both Mac OS X and Linux.

KathleenLabrie, 2022-09-28:
Forgot to be specific about the branch:  `release/3.1.x`

chris-simpson, 2022-09-28:
It appears to happen on `master` too

chris-simpson, 2022-09-29:
fix pushed to `bugfix/skysub_infinite_loop` which will need to be merged into both `master` and `release/3.1.x` if tests pass

chris-simpson, 2022-09-29:
tests pass, commit implemented in both branches, closing issue

chris-simpson, 2022-10-02:
I realize there are other edge cases which aren't handled properly so I'm reopening this

chris-simpson, 2022-10-04:
Should be OK now. There was an issue where all the skies were single frames and you were doing something like subtracting B from A and A from B where you can't sky-subtract either "in place" without messing up the other subtraction.

chris-simpson, 2022-10-04:
All good. Closing ticket again.

closed 2022-10-04 by chris-simpson