GeminiDRSoftware/DRAGONS #389: measureIQ hard crashes on GMOS LS 2D spectrum

KathleenLabrie, 2022-09-28:
Data:  https://drive.google.com/drive/folders/15O7xsJsagV8S-IHUijYFGuqatd0POE-E?usp=sharing

Run:  `reduce -r measureIQ N20180108S0053_aperturesFound.fits`

`release/3.1.x` latest pull (commit 58299cb206e26)

```
   PRIMITIVE: measureIQ
   --------------------
ERROR - Traceback (most recent call last):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/reduction/coreReduce.py", line 238, in runr
ERROR -     primitive_as_recipe()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 364, in gn
ERROR -     ret_value = fn(pobj, adinputs=adinputs, **fnargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 293, in gn
ERROR -     ret_value = fn(pobj, **kwargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 376, in measureIQ
ERROR -     report.report(results)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 1238, in report
ERROR -     if results.get("elip", 0) > 0.1:
ERROR - TypeError: '>' not supported between instances of 'NoneType' and 'float'
ERROR - '>' not supported between instances of 'NoneType' and 'float'
ERROR - reduce caught an unhandled exception.
ERROR - 
ERROR - '>' not supported between instances of 'NoneType' and 'float'
ERROR - Traceback (most recent call last):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/scripts/reduce", line 87, in main
ERROR -     r_reduce.runr()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/reduction/coreReduce.py", line 238, in runr
ERROR -     primitive_as_recipe()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 364, in gn
ERROR -     ret_value = fn(pobj, adinputs=adinputs, **fnargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 293, in gn
ERROR -     ret_value = fn(pobj, **kwargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 376, in measureIQ
ERROR -     report.report(results)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 1238, in report
ERROR -     if results.get("elip", 0) > 0.1:
ERROR - TypeError: '>' not supported between instances of 'NoneType' and 'float'
ERROR - 
ERROR - Reduce instance aborted.
```

chris-simpson, 2022-09-30:
Trivial fix. Just some bad coding that "elip" is set to `None` for spectra and so `results.get("elip", 0)` still returns `None` and not the desired float value of 0. Waiting to push pending completion of tests on previous commit.

chris-simpson, 2022-10-01:
Sorted.

closed 2022-10-01 by chris-simpson