GeminiDRSoftware/DRAGONS #390: measureIQ does not find nice source almost centered in the slit

KathleenLabrie, 2022-09-29:
Older versions were finding the source and measuring IQ.  In release/3.1.x, it claims no good sources.  See screenshot.
[image]

Data:  https://drive.google.com/drive/folders/1O0DyGs5PZ6ivXpOvhWu4CW6LGrE-A6HJ?usp=sharing
Run: `reduce -r measureIQ N20180521S0099_ADUToElectrons.fits`

chris-simpson, 2022-09-30:
Oops. This is a refactoring error on my part that accidentally missed out some functionality in an effort to improve efficiency. It's not trivial to sort out.

chris-simpson, 2022-10-01:
Fix and tidy in new branch `bugfix/measureiq_ls` from `master`. Waiting for tests and will cherry-pick into `release/3.1.x`

chris-simpson, 2022-10-04:
Now in both `master` and `release/3.1.x`. Tests pending.

chris-simpson, 2022-10-04:
Tests passed. Closing ticket.

closed 2022-10-04 by chris-simpson