GeminiDRSoftware/DRAGONS #395: addIllumMaskToDQ unreliable after recent large shift of the longlist in GMOS-S

KathleenLabrie, 2022-10-01:
The longslit in GMOS-S has shifted by a large amount following the recent work on the instrument.  With the B600 grating, in particular, the alignment of the illumination mask is failing which puts a serious limitation on the automation of the reduction (eg. with FIRE)

`release/3.1.x` and `release/3.0.x` fail in different ways.  They do use different algorithm.  The failure of 3.0.x is complete.  The failure of 3.1.x is partial once the `max_shift` is adjusted.

A science and flat sequence was found that illustrates many problems.  A spreadsheet with offset results with both branches and with various values of `max_shift`, along with the data are available at: 

https://drive.google.com/drive/folders/12fQ_mN5cPsN_uWqcXuG-0w6h8gxo3p2g?usp=sharing

(data is proprietary and access it limited to the DRAGONS team.)

In the current state with 3.0.x failing completely (even changing max_shift does not help), FIRE reduction of GMOS-S longslit ToO is somewhat compromised.

chris-simpson, 2022-10-06:
This is working in 3.0.x. I'm struggling with 3.1.x trying to get the new fixture to work. (This is to remind me of the status because there are so many things going on and I can't remember which fixes have been merged into which branches.)

chris-simpson, 2022-10-07:
3.1.x now picking up the new fixture and working; commits have been cherry-picked into master now, which *should* work, and then i can close this

chris-simpson, 2022-10-07:
All three branches are happy finally!

closed 2022-10-07 by chris-simpson