GeminiDRSoftware/DRAGONS #398: measureIQ messes up, somehow, such that the overscan trimming is not happening

KathleenLabrie, 2022-10-11:
If `measureIQ` is run before `overscanCorrect` (done in QAP), the trimming is not happening.  The logs claim that "No changes will be made to S20140623S0133_overscanSubtracted.fits extension 1, since the data section matches the data shape".  This seems to indicate that `measureIQ` is messing with DATASEC or something like that.

Data: https://drive.google.com/drive/folders/1N_PXX5H54yhuiLARBuylio3UGgNjYcTL?usp=sharing  

```
reduce -r measureIQ S20140623S0133_sourcesDetected.fits
reduce -r overscanCorrect S20140623S0133_iqMeasured.fits

echo -e "import astrodata\nimport gemini_instruments\nad=astrodata.open('S20140623S0133_overscanCorrected.fits')\nad.info()" | python

...
[ 0]   science                  NDAstroData       (4224, 544)    float32
...
```

```
reduce -r overscanCorrect S20140623S0133_sourcesDetected.fits

echo -e "import astrodata\nimport gemini_instruments\nad=astrodata.open('S20140623S0133_overscanCorrected.fits')\nad.info()" | python

...
[ 0]   science                  NDAstroData       (4224, 512)    float32
...
```

chris-simpson, 2022-10-11:
This is more blowback from our crazy desire to tile GMOS images with the overscan section still on, which dates back to when we used `gbias` to do the overscan correct, and that was slow. There will be a quick fix but I think the best solution is to stop this nonsense where there's a LUT with the estimated bias levels (which is often wrong) and to subtract those from the data and instead for `display` to properly `overscanCorrect` (a copy of) the image before displaying.

KathleenLabrie, 2022-10-11:
But if I display a raw frame, right now, I am seeing the overscan regions.  They are part of the data.  `overscanCorrect` would remove them.

Maybe you mean `overscanSubtract`?

chris-simpson, 2022-10-11:
That's my point. When you display a raw frame the overscan regions remain because it was originally considered too slow to remove them. Although I guess maybe we want them to remain in case there's a problem in those regions.

chris-simpson, 2022-10-11:
Should be fixed in `3.1.x`. `tileArrays()` has to modify the input if the overscan region is still present so needs to work on a `deepcopy` of the AD if called by `measureIQ()` or, indeed, `display()`. I also tidied up `display()` as it was deepcopying *twice* to add the DQ and also when subtracting the bias level. I now track whether it's working on a copy or the original and don't bother to `deepcopy` again.

KathleenLabrie, 2022-10-11:
It works... with lovely HELLO [Section... debugging statement.  :)

chris-simpson, 2022-10-11:
Oops. Removed (I squashed the commits and force-pushed so I only have to cherry-pick the one into master, which I've done). I'll close this.

closed 2022-10-11 by chris-simpson