GeminiDRSoftware/DRAGONS #399: measureIQ crash on processed data (after `traceApertures`)

KathleenLabrie, 2022-10-13:
Data: https://drive.google.com/drive/folders/1f3yh_X0VPpoq6rCl6IJF4Hla70EHYKCh?usp=sharing

To reproduce:  reduce -r measureIQ N20180108S0053_aperturesTraced.fits

Branch: release/3.1.x

Trace:
```
================================================================================
RECIPE: measureIQ
================================================================================
   PRIMITIVE: measureIQ
   --------------------
ERROR - Traceback (most recent call last):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/reduction/coreReduce.py", line 238, in runr
ERROR -     primitive_as_recipe()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 364, in gn
ERROR -     ret_value = fn(pobj, adinputs=adinputs, **fnargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 293, in gn
ERROR -     ret_value = fn(pobj, **kwargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 359, in measureIQ
ERROR -     nsources = report.add_measurement(ext, strehl_fn=partial(_strehl, ext))
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 1068, in add_measurement
ERROR -     else gt.fit_continuum(ext))
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/gempy/gemini/gemini_tools.py", line 1331, in fit_continuum
ERROR -     m_final = fit_it(m_init, pixels[mask], data[mask])
ERROR -   File "/Users/klabrie/condaenvs/susd3.9_20220816/lib/python3.9/site-packages/astropy/modeling/fitting.py", line 264, in wrapper
ERROR -     return func(self, model, x, y, z=z, **kwargs)
ERROR -   File "/Users/klabrie/condaenvs/susd3.9_20220816/lib/python3.9/site-packages/astropy/modeling/fitting.py", line 1159, in __call__
ERROR -     fitparams, cov_x, dinfo, mess, ierr = optimize.leastsq(
ERROR -   File "/Users/klabrie/condaenvs/susd3.9_20220816/lib/python3.9/site-packages/scipy/optimize/_minpack_py.py", line 414, in leastsq
ERROR -     raise TypeError(f"Improper input: func input vector length N={n} must"
ERROR - TypeError: Improper input: func input vector length N=5 must not exceed func output vector length M=2
ERROR - Improper input: func input vector length N=5 must not exceed func output vector length M=2
ERROR - reduce caught an unhandled exception.
ERROR - 
ERROR - Improper input: func input vector length N=5 must not exceed func output vector length M=2
ERROR - Traceback (most recent call last):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/scripts/reduce", line 87, in main
ERROR -     r_reduce.runr()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/reduction/coreReduce.py", line 238, in runr
ERROR -     primitive_as_recipe()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 364, in gn
ERROR -     ret_value = fn(pobj, adinputs=adinputs, **fnargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 293, in gn
ERROR -     ret_value = fn(pobj, **kwargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 359, in measureIQ
ERROR -     nsources = report.add_measurement(ext, strehl_fn=partial(_strehl, ext))
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 1068, in add_measurement
ERROR -     else gt.fit_continuum(ext))
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/gempy/gemini/gemini_tools.py", line 1331, in fit_continuum
ERROR -     m_final = fit_it(m_init, pixels[mask], data[mask])
ERROR -   File "/Users/klabrie/condaenvs/susd3.9_20220816/lib/python3.9/site-packages/astropy/modeling/fitting.py", line 264, in wrapper
ERROR -     return func(self, model, x, y, z=z, **kwargs)
ERROR -   File "/Users/klabrie/condaenvs/susd3.9_20220816/lib/python3.9/site-packages/astropy/modeling/fitting.py", line 1159, in __call__
ERROR -     fitparams, cov_x, dinfo, mess, ierr = optimize.leastsq(
ERROR -   File "/Users/klabrie/condaenvs/susd3.9_20220816/lib/python3.9/site-packages/scipy/optimize/_minpack_py.py", line 414, in leastsq
ERROR -     raise TypeError(f"Improper input: func input vector length N={n} must"
ERROR - TypeError: Improper input: func input vector length N=5 must not exceed func output vector length M=2
ERROR - 
ERROR - Reduce instance aborted.

reduce exit status: 6
```

chris-simpson, 2022-10-13:
This is an issue of a duff aperture failing to fit and not being picked up; I think it's just a case of catching the exception and moving on.

chris-simpson, 2022-10-13:
This specific error is because there are too few unmasked pixels compared to the number of degrees of freedom of the fit so I check for this first and carry on without trying to fit. But I've also wrapped the fit in a general try--except to not crash. Pushed to both `release/3.1.x` and `master.

chris-simpson, 2022-10-13:
I'm reopening this because the error on the measurement for this case is unhelpfully large. It's not clear whether it's the result of some resolved sources or bad apertures, but the rejection should be better I feel.

```
      FWHM measurement:                   2.528 +/- 1.444 arcsec
      Zenith-corrected FWHM (AM 1.78):    1.787 +/- 1.020 arcsec
```

chris-simpson, 2022-10-14:
No, this needs significant work. I'll write an issue explaining what might need doing and then close this again.

closed 2022-10-14 by chris-simpson