GeminiDRSoftware/DRAGONS #404: addDQ fails on a particular file that used to work

KathleenLabrie, 2022-10-25:
Branch:  release/3.0.x     (note that it fails in a much more scary way in 3.1.x, so probably easier to troubleshoot in 3.0.x)

Data: S20140718S0030.fits

To reproduce:
`reduce -r addDQ S20140718S0030.fits`

Trace: 
```
   PRIMITIVE: addDQ
   ----------------
   Clipping gmos-s_bpm_HAM_42_full_12amp_v1.fits to match science data.
ERROR - Traceback (most recent call last):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/reduction/coreReduce.py", line 224, in runr
ERROR -     primitive_as_recipe()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 334, in gn
ERROR -     ret_value = fn(pobj, adinputs=adinputs, **fnargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/core/primitives_standardize.py", line 105, in addDQ
ERROR -     ext.mask |= static_ext.data
ERROR - ValueError: operands could not be broadcast together with shapes (2112,160) (2112,32) (2112,160)
ERROR - operands could not be broadcast together with shapes (2112,160) (2112,32) (2112,160) 
ERROR - reduce caught an unhandled exception.
ERROR - 
ERROR - operands could not be broadcast together with shapes (2112,160) (2112,32) (2112,160) 
ERROR - Traceback (most recent call last):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/scripts/reduce", line 92, in main
ERROR -     r_reduce.runr()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/reduction/coreReduce.py", line 224, in runr
ERROR -     primitive_as_recipe()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 334, in gn
ERROR -     ret_value = fn(pobj, adinputs=adinputs, **fnargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/core/primitives_standardize.py", line 105, in addDQ
ERROR -     ext.mask |= static_ext.data
ERROR - ValueError: operands could not be broadcast together with shapes (2112,160) (2112,32) (2112,160)
ERROR - 
ERROR - Reduce instance aborted.

reduce exit status: 6

```

I checked the file, it is a solid 2112,160 for all extensions.  I don't know where that 2112, 32 is coming from.

chris-simpson, 2022-10-31:
I'm certain I've seen this problem before. It's a mangled header in the "science" frame. Here's what `array_section()` returns:
```
[1:512,1:4224]
[513:1024,1:4224]
[1025:1536,1:4224]
[1537:2048,1:4224]
[1537:2048,1:4224]
[2049:2560,1:4224]
[2561:3072,1:4224]
[3073:3584,1:4224]
[3073:3584,1:4224]
[3585:4096,1:4224]
[4097:4608,1:4224]
[4609:5120,1:4224]
```
and here's what it _should_ return:
```
[1:512,1:4224]
[513:1024,1:4224]
[1025:1536,1:4224]
[1537:2048,1:4224]
[1:512,1:4224]
[513:1024,1:4224]
[1025:1536,1:4224]
[1537:2048,1:4224]
[1:512,1:4224]
[513:1024,1:4224]
[1025:1536,1:4224]
[1537:2048,1:4224]
```
(which is also what the BPM has) so there are no array sections in the BPM to match those in the science (after the first few extensions) and it simply creates an overscan region (32 pixels wide) and attaches it to nothing. Not the most elegant failure but it won't work.

I'm also certain it didn't work the last time you ran the tests as I have seen this exact problem before (although I cannot swear it was the same file).

KathleenLabrie, 2022-11-10:
The bad CCDSEC stopped in mid-August 2014.  I was most likely related to the commissioning of the Ham CCDs which started in June 2014.   I have a long list of files that Oliver found.  I’ll see if there’s a pattern, ie. why some data are affected, some are not.  Then we will have to decide whether we want DRAGONS (in prepare?) to handle those cases, or if we want to fix the files in the archive.   (Most likely "fix files in archive" would be the best solution but let's inspect the files first.)

Closing ticket.

closed 2022-11-10 by KathleenLabrie