GeminiDRSoftware/DRAGONS #405: measureIQ failure after trace (release/3.0.x)

KathleenLabrie, 2022-10-25:
Branch: release/3.0.x
Data: https://drive.google.com/drive/folders/1c6F4vMtMGxXP5FEAd3Xliz35C7MW6hWp?usp=sharing

Python 3.7 env, so older astropy dependencies.  (I have a hunch that this one is related to version mismatch.)   

Run:
   `reduce -r measureIQ N20180108S0053_aperturesTraced.fits`

Trace:
```
   PRIMITIVE: measureIQ
   --------------------
ERROR - Traceback (most recent call last):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/reduction/coreReduce.py", line 224, in runr
ERROR -     primitive_as_recipe()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 334, in gn
ERROR -     ret_value = fn(pobj, adinputs=adinputs, **fnargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 513, in measureIQ
ERROR -     gt.fit_continuum(adiq)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/gempy/gemini/gemini_tools.py", line 1155, in fit_continuum
ERROR -     m_final = fit_it(m_init, pixels[mask], data[mask])
ERROR -   File "/Users/klabrie/condaenvs/susd3.7_3.0.x_20210907/lib/python3.7/site-packages/astropy/modeling/fitting.py", line 261, in wrapper
ERROR -     return func(self, model, x, y, z=z, **kwargs)
ERROR -   File "/Users/klabrie/condaenvs/susd3.7_3.0.x_20210907/lib/python3.7/site-packages/astropy/modeling/fitting.py", line 1159, in __call__
ERROR -     xtol=acc, full_output=True)
ERROR -   File "/Users/klabrie/condaenvs/susd3.7_3.0.x_20210907/lib/python3.7/site-packages/scipy/optimize/minpack.py", line 414, in leastsq
ERROR -     raise TypeError(f"Improper input: func input vector length N={n} must"
ERROR - TypeError: Improper input: func input vector length N=5 must not exceed func output vector length M=2
ERROR - Improper input: func input vector length N=5 must not exceed func output vector length M=2
ERROR - reduce caught an unhandled exception.
ERROR - 
ERROR - Improper input: func input vector length N=5 must not exceed func output vector length M=2
ERROR - Traceback (most recent call last):
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/scripts/reduce", line 92, in main
ERROR -     r_reduce.runr()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/reduction/coreReduce.py", line 224, in runr
ERROR -     primitive_as_recipe()
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/recipe_system/utils/decorators.py", line 334, in gn
ERROR -     ret_value = fn(pobj, adinputs=adinputs, **fnargs)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/geminidr/gemini/primitives_qa.py", line 513, in measureIQ
ERROR -     gt.fit_continuum(adiq)
ERROR -   File "/Users/klabrie/data/GemKLRepo/activedragons/gempy/gemini/gemini_tools.py", line 1155, in fit_continuum
ERROR -     m_final = fit_it(m_init, pixels[mask], data[mask])
ERROR -   File "/Users/klabrie/condaenvs/susd3.7_3.0.x_20210907/lib/python3.7/site-packages/astropy/modeling/fitting.py", line 261, in wrapper
ERROR -     return func(self, model, x, y, z=z, **kwargs)
ERROR -   File "/Users/klabrie/condaenvs/susd3.7_3.0.x_20210907/lib/python3.7/site-packages/astropy/modeling/fitting.py", line 1159, in __call__
ERROR -     xtol=acc, full_output=True)
ERROR -   File "/Users/klabrie/condaenvs/susd3.7_3.0.x_20210907/lib/python3.7/site-packages/scipy/optimize/minpack.py", line 414, in leastsq
ERROR -     raise TypeError(f"Improper input: func input vector length N={n} must"
ERROR - TypeError: Improper input: func input vector length N=5 must not exceed func output vector length M=2
ERROR - 
ERROR - Reduce instance aborted.

```

The input file used an arc created with asdf-2.9.0.dev0 and gwcs-0.16.1 but the 3.0.x env has asdf-2.7.1 and gwcs-0.14.0.   (The calibrations were created with v3.1 and are being used for 3.0.  I'm testing after all.  Most of the time it works btw.)   

I would like 
1) confirmation that the version discrepancies are the cause of the crash.
2) Some understanding of how the reverse is expected to work.  Eg. calibrations in archive reduced with older version of asdf and gwcs and astropy, used with newer version of the software.   
3) why in the world should measureIQ care at all about asdf and gwcs if it's just collapsing sections of the 2D spectrum and measuring fwhm of the cross section?

chris-simpson, 2022-10-31:
This is just #399 isn't it, that needs merging into 3.0.x?

chris-simpson, 2022-10-31:
Yes it is. I've cherry-picked that commit.

closed 2022-10-31 by chris-simpson