GeminiDRSoftware/DRAGONS #437: ad[0].multiply(1.) fails with astropy 5.3

KathleenLabrie, 2023-06-03:
The astrodata multiply method fails with astropy 5.3.   It does not fail on a raw file, so it has to be something to do with either the variance or the mask.

Test file at: https://drive.google.com/drive/folders/1diCPWxBYZ7Dw6SHmGyJkJ2Z3IHxvtyxZ?usp=share_link

```
import astrodata, gemini_instruments

ad = astrodata.open('N20160102S0423_varAdded_crash.fits')
ad[0].multiply(1.)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/klabrie/data/GemKLRepo/activedragons/astrodata/core.py", line 798, in __imul__
    self._standard_nddata_op(NDDataObject.multiply, oper)
  File "/Users/klabrie/data/GemKLRepo/activedragons/astrodata/core.py", line 759, in _standard_nddata_op
    return self._oper(partial(fn, handle_mask=np.bitwise_or,
  File "/Users/klabrie/data/GemKLRepo/activedragons/astrodata/core.py", line 756, in _oper
    ndd[ind[n]] = operator(ndd[ind[n]], operand)
  File "/Users/klabrie/condaenvs/testastropy53/lib/python3.10/site-packages/astropy/nddata/mixins/ndarithmetic.py", line 618, in multiply
    return self._prepare_then_do_arithmetic(
  File "/Users/klabrie/condaenvs/testastropy53/lib/python3.10/site-packages/astropy/nddata/mixins/ndarithmetic.py", line 731, in _prepare_then_do_arithmetic
    result, init_kwds = operand._arithmetic(operation, operand2, **kwargs)
  File "/Users/klabrie/data/GemKLRepo/activedragons/astrodata/nddata.py", line 73, in _arithmetic
    return super()._arithmetic(
  File "/Users/klabrie/condaenvs/testastropy53/lib/python3.10/site-packages/astropy/nddata/mixins/ndarithmetic.py", line 335, in _arithmetic
    kwargs["mask"] = self._arithmetic_mask(
  File "/Users/klabrie/condaenvs/testastropy53/lib/python3.10/site-packages/astropy/nddata/mixins/ndarithmetic.py", line 527, in _arithmetic_mask
    return handle_mask(self.mask, operand.mask, **kwds)
TypeError: unsupported operand type(s) for |: 'int' and 'NoneType'
```

saimn, 2023-06-05:
Probably related to https://github.com/astropy/astropy/pull/14175

KathleenLabrie, 2023-06-21:
Also broken is the multiplication with a file without a mask.

`ad_withmask.multiply(ad_nomask)`

If I do `ad_withmask.multiply(ad_withmask)`, no error.   Starting to look more and more like a bug in astropy itself.   The propagation of the mask does not take into account that one of the operand might not have a mask.

KathleenLabrie, 2023-06-22:
Completely removing `astrodata` from the equation.  Just using `NDDataRef` now.   Either I'm not using it the right way, or there is a bug in `astropy` v5.3.   I need someone with more knowledge than me to assess that.

```
>>> import numpy as np
>>> from astropy.nddata import NDDataRef

>>> array = np.array([[0, 1, 0], [1, 0, 1], [0, 1, 0]])
>>> mask = np.array([[0, 1, 64], [8, 0, 1], [2, 1, 0]])

>>> nref_nomask = NDDataRef(array)
>>> nref_mask = NDDataRef(array, mask=mask)

# multiply no mask by constant (no mask * no mask)
>>> nref_nomask.multiply(1., handle_mask=np.bitwise_or).mask   # returns nothing, no mask

# multiply no mask by itself (no mask * no mask)
>>> nref_nomask.multiply(nref_nomask, handle_mask=np.bitwise_or).mask # return nothing, no mask

# multply mask by constant (mask * no mask)
>>> nref_mask.multiply(1., handle_mask=np.bitwise_or).mask
...
TypeError: unsupported operand type(s) for |: 'int' and 'NoneType'

# multiply mask by itself (mask * mask)
>>> nref_mask.multiply(nref_mask, handle_mask=np.bitwise_or).mask
array([[ 0,  1, 64],
       [ 8,  0,  1],
       [ 2,  1,  0]])

# multiply mask by no mask (mask * no mask)
>>> nref_mask.multiply(nref_nomask, handle_mask=np.bitwise_or).mask
...
TypeError: unsupported operand type(s) for |: 'int' and 'NoneType'
```

@chris-simpson @saimn, do you think that this needs reporting to `astropy` or am I doing something I shouldn't be doing?

saimn, 2023-06-22:
@KathleenLabrie - Yes please open a new issue, with the example to reproduce :+1:

KathleenLabrie, 2025-04-17:
Not an issue anymore in astropy v7.  Closing.

closed 2025-04-17 by KathleenLabrie