GeminiDRSoftware/DRAGONS #465: Parameter mismatch in adjustWCSToReference

nguevara, 2024-09-13:
Hi, I just wanted to point out that in v3.2.0 and v3.2.1 there's a mismatch between the `adjustWCSToReference` primitive parameters listed in `/geminidr/core/primitives_spect.py` and the parameter file `/geminidr/core/parameters_spect.py`.
The first one says:
```
 method : str ['correlation' | 'offsets']
            Method to use to compute offsets. 'correlation' uses a
            correlation of the slit profiles (the 2d images stacked
            on the dispersion axis), 'offsets' uses the QOFFSET keyword.
```
while the second one says:
```
    method = config.ChoiceField("Alignment method", str,
                                allowed={"sources_wcs": "Match sources using WCS",
                                         "sources_offsets": "Match sources using telescope offsets",
                                         "offsets": "Use telescope offsets only"},
                                default="sources_wcs", optional=False)
```

KathleenLabrie, 2025-03-06:
Thanks for reporting.  Much appreciated.  I just fixed it in the code base on the release/3.2.x branch.

Apologies for taking so long to respond.

KathleenLabrie, 2025-03-06:
Changes made in `release/3.2x`.  Will be merged to main branch later.

closed 2025-03-06 by KathleenLabrie