GeminiDRSoftware/DRAGONS #474: PackagesNotFoundError on install

danielhill676, 2025-03-03:
With correct channels, calling conda create -n dragons python=3.10 dragons ds9 returns:

Channels:
 - http://astroconda.gemini.edu/public
 - conda-forge
 - http://ssb.stsci.edu/astroconda
 - defaults
Platform: osx-arm64
Collecting package metadata (repodata.json): done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - ds9
  - dragons

Current channels:

  - http://astroconda.gemini.edu/public
  - https://conda.anaconda.org/conda-forge
  - http://ssb.stsci.edu/astroconda
  - defaults

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

Is this a common error?

teald, 2025-03-03:
Hi @danielhill676, you need to use osx-64, not osx-arm64 (looking at the `Platform:` output above). Please see the [Installation Guide](https://dragons.readthedocs.io/projects/recipe-system-users-manual/en/v3.2.0/install.html), specifically the warning in "[Download and Install Miniforge](https://dragons.readthedocs.io/projects/recipe-system-users-manual/en/v3.2.0/install.html#download-and-install-miniforge)".

KathleenLabrie, 2025-03-06:
Also, the ssb.stsci.edu/astroconda channel should be removed from your list of channels.  STScI has stopped support.  I'm actually surprised that that didn't issue an error or a warning, I thought it had been archived.

aarranshaw, 2025-04-15:
Piggybacking on this issue instead of submitting a new one as I had the same problem with platforms. I already have osx-arm64 miniforge (mamba, to be specific) for all of my other python environments. Is there a safe way of installing a completely separate osx-64 miniforge that I would only activate for using DRAGONS? I would like to keep osx-arm64 for my other environments, but I also want to reduce and analyze my Gemini data!