GeminiDRSoftware/DRAGONS #477: Left menu options in interactive findApertures are not working

KathleenLabrie, 2025-03-06:
In `release/3.2.x`, some options in the left-top section of the interactive tool for findApertures are broken.  Changing them are recalculating has absolutely no effect.   Changing the same parameter on the command line does work, so it isn't the code associated with the parameter settings that doesn't work, it appears to be that the interactive tool does not apply the changes requested from within the tool.

**Currently identified as broken are:**
- "Use S/N ratio in spatial profile?" checkbox.  Turning it off ("on" by default) and clicking "Find apertures" changes nothing.  (I have verified that setting it off in the call rather than in the tool does make a big difference.)
- "Pixel section(s) ..." box is also unresponsive.  (Again, I have checked that setting it out of the interactive tool does bring a change.)

**Minor issue:** 
- The "Threshold" slider is supposed to have a range from 0 to 1, but I can slide it to larger number (then get a pop saying max 1)

**Verified to be working:**
- Percentile
- SNR threshold
- Max apertures
- Maximum separation from target

**Unable to verify:**
- Min sky region

----

Regarding the broken items, 

**The S/N checkbox** 
- used to work in v3.1 which is bokeh 2.4.  (v3.2 is bokeh >3).  So this must have been introduced during the refactoring.   
- On the master branch, it has an effect but not the right one and it is probably more an algorithm issue rather than a GUI issue.   (I'll investigate further and either update here or create a new ticket.)

**The "Pixel section" box** 
- appears to be broken in v3.1 too, so it has been broken for a while.
- works fine on the master branch.

----
The command to reproduce:
```
reduce -r findApertures N20240817S0047_stack.fits -p interactive=True
```
(To see what the broken options do when called from the command line, add `-p section=1200:1300` for pixel section, and `-p use_snr=False` for the S/N checkbox.)

The stack is from proprietary and can only be shared within NOIRLab for debugging purposes (the link only works for NOIRLab staff).  It can be found here: 

[https://drive.google.com/drive/folders/1Uwx2HDBVxk9G1JMjaDT7Zxhs1fG1H-9f?usp=drive_link](https://drive.google.com/drive/folders/1Uwx2HDBVxk9G1JMjaDT7Zxhs1fG1H-9f?usp=drive_link)
----

**I recommend:**

- that the S/N checkbox issue be fixed in release/3.2.x because in the master branch there seems to be another issue mixed in that will just cause more confusion.  
- that the Pixel section issue be left alone since it works in the master branch and our next release will be off that branch.  Might good to understand why it doesn't work in 3.2.x though.

KathleenLabrie, 2025-03-06:
@chris-simpson @teald

teald, 2025-03-06:
I'll take a look at this later today.

teald, 2025-03-07:
I just tried to reproduce the behavior, and got some (maybe) weirder behavior.

I'm running this with `release/3.2.x` and a fresh development environment (directly from the README). This is using Firefox for the browser. (I got the same thing on safari unless otherwise noted, though)

## S/N checkbox

When I uncheck this box and click Find Apertures my apertures disappear completely. Is this is the behavior you also got before/what you experienced with `master`? What browser are you using?

When I re-check the box, I get different apertures from the original solution (same on `master`, as well, though it picks up another aperture at 2030 in the "original"). If I hit reset it goes back to the original. I'll send screenshots directly to you (since the data is proprietary).

## Pixel sections

I'm also not getting any updates on this on 3.2.x, but seems to work for me on `master` as well. I'll check for differences and see if the fix was straightforward.

## Threshold

I'm also able to reproduce this on `release/3.2.x`. I vaguely recall this being an issue before, and it works on `master`. Same as above, I'll check if the fix was a straightforward one or not.

teald, 2025-03-20:
If I hit reset after unchecking and rechecking S/N Ratio (and finding apertures each step there), I go back to the original solution. I only get the different solution if I uncheck, find apertures, recheck, and find apertures again.